antiprism/mpd_oled #73: MPD OLED for Raspberry Pi VOLUMIO 3.0?

Korniman-the-Rooky, 2021-10-30:
Dear Adrian, do you will create this "best of it's art plugin" MPD OLED also for VOLUMIO 3.0 running on Raspberry Pi?
Kind Regards
Benjamin

antiprism, 2021-10-30:
Hi Benjamin

I imagine that the mpd_oled program already run on Volumio 3, but that the install instructions will need updating and a new script  will be needed to set up a copy of the audio for the spectrum analyser. My understanding is that the audio copy is supposed to be made through a Volumio 3 plugin API which hasn't yet been fully documented, so I'll review this when Volumio 3 is closer to release. Also, if @supercrab is interested in converting the mpd_oled Volumio 2 plugin to Volumio 3 then I will look at what I can do to support that.

I'll leave the issue open to monitor the situation. Thanks for your interest.

Adrian.

Korniman-the-Rooky, 2021-10-30:
Dear Adrian, thank you for your prompt reply. Yes your plugin with the
supercrap plugin routine and GUI works perfect. Ok then let's wait when V
3.0 is officially released.
Kind regards

Mit freundlichen Grüßen
Benjamin Korntheuer

supercrab, 2021-10-30:
Hi guys!

I made a version of the plugin for volumio 3.0 but I couldn’t release it because the audio routing wouldn’t work. It did work for others so maybe it’s down to my sound device?  When volumio 3 is released I’ll have another go at making it work! 👍🏽👍🏽😃

antiprism, 2021-10-31:
Hi Mase

Great that the plugin is working apart from issues with the audio routing.

From experiences on Moode, which will now insert an audio copy plugin device in the ALSA configuration, there are likely to be issues depending on where 'pluginhw' and 'hw' are used in the overall ALSA configuration. In particular, if a user can insert extra ALSA plugin devices in the audio configuration through enabling Volumio 3 plugins (e.g. an equaliser) then the presence of the audio copy plugin device could cause playback to fail, or could lock the playback parameters (e.g. bitrate) to the values being used when cava was first started by mpd_oled.

Some of these issues might be fixed if there was an option for mpd_oled to stop cava when the player state changed to 'stop', and start cava again whn the player state changed to 'play/pause', but if the change of player state moving from play->stop->play was less than 0.1 seconds, e.g. when playing a playlist, it is possible that cava would not be restarted (potentially causing a "random" issue from the point of view of the user). There would also be a an adjustment of the spectrum levels at each new 'play', but this is now only a few seconds, and is acceptable to me, but could be annoying for some people or in some situations.

Adrian.

supercrab, 2021-11-01:
I was looking at this whole thing a few months ago as soon as Volumio 3 dev version came out.   I think someone commented on my post in the forum about the plugin.  After hours of trying to get the audio sent to the analyser I gave up.

I can spend hours coding in C++, Javascript, C# but getting the ALSA config/audio routing working is something else.  I'd rather put a fire out with my face.  

When Volumio 3 is out I'll take another bash though as the version stable  

There's a new Pi Zero 0 - I snapped one up as soon as I heard about them.

Also, how is the new version going with the new graphics library?

antiprism, 2021-11-02:
Hi Mase

The new Pi Zero W 2 looks perfect for mpd_oled. The CPU usage sometimes touches 100% for me on a Pi Zero, and the extra processing power with the new version should really help smooth the spectrum and avoid occaisional audio stutter.

I haven't had much opportunity to look at the development version of mpd_oled recently, and so no updates to mpd_oled_dev since July!

Adrian.

alain93, 2021-12-07:
Hi Adrian,
Any news for V3? I know the it's working with V3 Beta, so it should work with V3.

alain93, 2021-12-07:
I just installed from source in the new V3 and it's working well.

antiprism, 2021-12-08:
Hi Alain

Thanks for confirming that mpd_oled is working on the release version of Volumio 3!

Adrian.

supercrab, 2021-12-08:
> I just installed from source in the new V3 and it's working well.

Awesome!  What audio routing settings did you use?

alain93, 2021-12-08:
Hi Supercrab,
Do you mean this? -o 6 -b 30 -g 1 -f 25 -P s
it's what I used. I didn't install your plugin.

supercrab, 2021-12-08:
> Hi Supercrab,
> Do you mean this? -o 6 -b 30 -g 1 -f 25 -P s
> it's what I used. I didn't install your plugin.

Not so much the the command line settings but how did you do setup the audio routing?  Did you use the Adrian’s setup script or modify any files manually?

alain93, 2021-12-08:
I followed the instructions in [Install instructions for Volumio 2 using
source](
https://github.com/antiprism/mpd_oled/blob/master/doc/install_volumio2_source.md
)
Didn't change anything.

supercrab, 2021-12-08:
Thank sir!

angeloschat, 2022-01-05:
One more vote for Volumio 3 .any updates?

angeloschat, 2022-01-06:
I installed from source on volumio 3 with Pi 0  2
All seems to work except the bars
Also complains about FIFO
`rm: cannot remove '/tmp/mpd_oled_fifo': No such file or directory`

antiprism, 2022-01-06:
Hi Angelo

I recommand rebooting your Pi, if you didn't already do this. If the bars still don't show when using an MPD source (music files, radio stations), please post the output from the following commands
```
cat /etc/mpd.conf
cat /volumio/app/plugins/music_service/mpd/mpd.conf.tmpl
ls /tmp/*fifo*
sudo systemctl status mpd_oled | cat
```

The 'rm' error can be ignored.

Adrian.

P.S. Just to note that even if the Volumio 2 audio copy method with an MPD FIFO still works on Volumio 3, it was always a hack and I won't be including it when I update the instructions for Volumio 3.

Korniman-the-Rooky, 2022-03-16:
Hi Adrian, Now I will change to VOLUMIO 3 and want to ask if the plugin is working there finally or if I have to modify?
@supercrab , Will your MPD OLED project with complete GUI support for VOLUMIO 2 work for VOLUMIO 3
Thank you for your feedback and have a nice day
Cheers Benjamin

antiprism, 2022-03-16:
Hi Benjamin

The mpd_oled program runs fine on Volumio 3 but, as ever, it may be a problem to set up the audio copy.

There is a version of Mase's plugin for Volumio 3, prepared by balbuze, that may or may not work depending on your system. The old method for making a copy of the audio, that was used with Volumio 2, still works, but will only work when using the MPD player (e.g. it won't work with Spotify). I recommend looking at the mpd_oled plugin thread on the Volumio forum

   https://community.volumio.org/t/plugin-mpd-oled-installation-configuration-plugin/44823/226

Adrian.

Korniman-the-Rooky, 2022-03-18:
Hi Andre, ok understands. Only for clarification, with volumio 2 in the past I had with spotify connect no equalizer bars, but title was visible on OLED and music was playing. Would it be the same error for VOLUMIO 3 (Musics plays and OLED shows title)?
If so, that would be ok for me
Kind Regards
Benjamin

supercrab, 2022-03-18:
+1 to what Aid said. The plugin only works well with Volumio 2 😪  I never managed to get it working with Volumio 3. 😢

Korniman-the-Rooky, 2022-03-18:
> +1 to what Aid said. The plugin only works well with Volumio 2 😪 I never managed to get it working with Volumio 3. 😢. Everyone else managed to get balbuze’s version working.

@supercrab , Hi Maze maybe I am stupid here, I was searching now 30 minutes for balbuze's version in the web. No results. Could you bring me to the next level?
cheers
Benjamin

antiprism, 2022-03-18:
Hi Benjamin

Maybe this is a working version of the Volumio 3 mpd_oled plugin

   https://community.volumio.org/t/not-recomended-plugin-and-tweaks-adventures-on-buster-betas/45552/57

Adrian.

Korniman-the-Rooky, 2022-03-18:
Ahh ok thank you Adrian!

Astronautra, 2022-05-21:
I got this green SSD1309 SPI display working this morning and I am LOVING IT!  
Only using it to play off a 2.5" drive with Volumio3.
Adrian.  Thank you so much for your instructions!!!  YOU ROCK!!!

[GreenMachine](https://user-images.githubusercontent.com/7470814/169661575-c5f9bdc5-623e-4fdd-8394-3114d09ccb7b.jpg)
!

antiprism, 2022-05-22:
Hi Astronautra

Looking good! Great that it worked out.

Adrian..

supercrab, 2022-08-08:
Am still working on getting the plugin to work on Volumio 3.  I thought it would be straight forward but there appear to be a few issues still but once I get these finished it should be able to included in Volumio

supercrab, 2022-11-01:
This is looking like it could be ready for V3 now and official release!

supercrab, 2022-11-04:
This issue can be closed now @antiprism 😎✅

supercrab, 2022-11-04:
> @supercrab , Hi Maze maybe I am stupid here, I was searching now 30 minutes for balbuze's version in the web. No results. Could you bring me to the next level? cheers Benjamin

@Astronautra  the plugin has been successfully merged into the volumio project

Astronautra, 2022-11-04:
> 

> @Astronautra the plugin has been successfully merged into the volumio project

...meaning it's an official Volumio Plug-In now?   Or just reporting success with the script here on 3.0 ?  Thanks for the update.

supercrab, 2022-11-04:
The code for the mpd_oled Volumio 3 plugin that I wrote has merged into the Volumio 3 repository.  The beta stable version of the plugin is already available for testing.  I can't remember how to install plugins in beta!

I'm not sure when the plugin will be official but the ball is rolling!

Astronautra, 2022-11-05:
> I'm not sure when the plugin will be official but the ball is rolling!
@supercrab   Just installed from the plugin inside Volumio interface.  Worked 1st try [SPI 1309 set to adafruit... all default ok]
[image]

antiprism, 2022-11-05:
Hi Mase

Mission accomplished. Well done!

Closing the issue.

Adrian.

closed 2022-11-05 by antiprism

supercrab, 2022-11-05:
Yay! 🥳🍾🎊🎉

2 years later 😳

supercrab, 2022-11-05:
> I'm not sure when the plugin will be official but the ball is rolling!
> @supercrab Just installed from the plugin inside Volumio interface. Worked 1st try [SPI 1309 set to adafruit... all default ok]
> [image]

Great news indeed! Thanks

antiprism, 2024-10-23:
The cava process (mpd_oled_cava) might have crashed.
Something might have gone wrong relating to the plugin's ALSA coniguration.

I recommend asking on the Volumio forum in the thread for the mpd_oled plugin: https://community.volumio.com/t/plugin-mpd-oled-installation-configuration-plugin/44823

Adrian.

supercrab, 2024-10-23:
Weird issue that!  It does sound familiar, although I don't use the `mpd_oled` plugin on Volumio 3 myself.  I've not heard of anyone else with the same issue.  When the ASLA stuff was implemented in Volumio 3 the plugin never worked properly for me, so I never able to properly support it :(