antiprism/mpd_oled #74: Display freezes on playback start (moOde 8.1.2)

merlin699, 2022-08-05:
I didn't use the script for a while becaus of a hardware change. Now I was coming back and wanted to use it again with my old hardware and the new 8.1.2 installed from scratch (using source, since the script didn't work for me).
In first place I get an error:
`E: Unable to locate package libi3c-dev`
when trying to install
`sudo apt install autoconf make libtool libfftw3-dev libiniparser-dev libmpdclient-dev libi3c-dev i2c-tools lm-sensors`
this might be the cause to freeze the display when using cava.
Continuing the install process runs through without problems and the display works fine as long it's on the "idle"-screen.
On starting playback the display switches to playback mode (trying to uses cava) the display freezes (no spectrum, song name, bitrate, time, etc..) and stays in this status.

antiprism, 2022-08-05:
Hi

Thanks for reporting the issue. The error relating to 'libi3c-dev' is an error in the documentation, having i3c instead of i2c!. I'll fix that.

Regarding the freeze, could you restart your player and run the command below twice, and post the output. Run it once before playing anything, and then again after playback starts and the screen is frozen
```
sudo systemctl status mpd_oled | cat
```
Adrian.

merlin699, 2022-08-05:
Hi Adrian,
thanks for your response.

The first output:

```
● mpd_oled.service - MPD OLED Display
     Loaded: loaded (/etc/systemd/system/mpd_oled.service; enabled; vendor preset: enabled)
     Active: active (running) since Fri 2022-08-05 15:47:37 CEST; 1min 38s ago
    Process: 434 ExecStartPre=/usr/bin/mpd_oled_launch_test (code=exited, status=0/SUCCESS)
   Main PID: 1173 (mpd_oled)
      Tasks: 2 (limit: 1010)
        CPU: 8.213s
     CGroup: /system.slice/mpd_oled.service
             └─1173 /usr/bin/mpd_oled -o 1 -b 15 -g 1 -f 25 -c alsa,hw:Loopback,1

Aug 02 22:32:26 moode-black systemd[1]: Starting MPD OLED Display...
Aug 05 15:47:37 moode-black systemd[1]: Started MPD OLED Display.

```

the second:

```
● mpd_oled.service - MPD OLED Display
     Loaded: loaded (/etc/systemd/system/mpd_oled.service; enabled; vendor preset: enabled)
     Active: active (running) since Fri 2022-08-05 15:47:37 CEST; 4min 21s ago
    Process: 434 ExecStartPre=/usr/bin/mpd_oled_launch_test (code=exited, status=0/SUCCESS)
   Main PID: 1173 (mpd_oled)
      Tasks: 3 (limit: 1010)
        CPU: 29.533s
     CGroup: /system.slice/mpd_oled.service
             └─1173 /usr/bin/mpd_oled -o 1 -b 15 -g 1 -f 25 -c alsa,hw:Loopback,1

Aug 02 22:32:26 moode-black systemd[1]: Starting MPD OLED Display...
Aug 05 15:47:37 moode-black systemd[1]: Started MPD OLED Display.
Aug 05 15:51:49 moode-black mpd_oled[6020]: Segmentation fault
```

antiprism, 2022-08-05:
Thanks. I'll install the latest Moode and try the binary and source mpd_oled installs and see if I have the same issues.

Adrian.

merlin699, 2022-08-05:
Thanks for your time, very appreciated.

antiprism, 2022-08-05:
Hi

I installed Moode 8.1.2 legacy on a Pi 4 with an SPI SSH1106, and enabled the headphone jack.

The mpd_oled binary package install went fine, with a working spectrum.

The source install, besides the package name correction, needed a couple of other packages installed which Moode no longer installs by default (git, libasound2-dev). I have updated the documentation, and with the extra packages installed the mpd_oled install from source went fine. However, I notice that the cava spectrum levels are low compared to the cava version in the binary package.

In the status output you sent the program being used is /usr/bin/mpd_oled, which is likely the version from the binary package. What went wrong when you installed the binary package?

Are you running the Moode legacy version or the 64-bit version? (I'll check the 54-bt version tomorrow anyway).

Adrian.

merlin699, 2022-08-06:
Hi,
thanks for your reply.
I'm using the 64 bit version (with a spi display, if that matters)
The installation went fine but the display didn't show anything. It stayed dark.

antiprism, 2022-08-06:
I checked this morning on 64-bit Moode 8.1.2 with an SPI display, and both the package and source (with the updated instructions) mpd_oled installations went fine, with a working spectrum display.

I had a thought, which is that when you were installing mpd_oled from source the install of cava may have completed (if you installed git), and the 32-bit mpd_oled binary from the package could be using the 64-bit mpd_oled_cava binary that you built. I am not sure about the compatibility between 64-bit and 32-bit here, but the mpd_oled_cava binary will not work correctly anyway because an essential package (libasound2-dev) was not installed when it was built. In which case, try to delete this binary
```
sudo rm /usr/local/bin/mpd_oled_cava
```
If you get a message saying `rm: cannot remove '/usr/local/bin/mpd_oled_cava': No such file or directory`, then that wasn't the problem, otherwise, the file was deleted and this may fix your issue.

Perhaps you could check playback using Radio 1, as this was my test source. If mpd_oled still crashes when you start playback then, perhaps you could say what audio hardware you are using?

To make testing easier, after mpd_oled has crashed, you can restart it with
```
sudo systemctl restart mpd_oled
```

Adrian.

merlin699, 2022-08-06:
Hi Adrian,
thanks for your work.
I tried
`sudo rm /usr/local/bin/mpd_oled_cava`
and indeed I got
`rm: cannot remove '/usr/local/bin/mpd_oled_cava': No such file or directory`
Then I made a fresh install with 8.1.2 and the binary version. The display worked after installing, but it crashed again when starting the playback.
```
● mpd_oled.service - MPD OLED Display
     Loaded: loaded (/etc/systemd/system/mpd_oled.service; enabled; vendor preset: enabled)
     Active: active (running) since Sat 2022-08-06 21:07:32 CEST; 2min 23s ago
    Process: 409 ExecStartPre=/usr/bin/mpd_oled_launch_test (code=exited, status=0/SUCCESS)
   Main PID: 1186 (mpd_oled)
      Tasks: 3 (limit: 1010)
        CPU: 2min 20.044s
     CGroup: /system.slice/mpd_oled.service
             └─1186 /usr/bin/mpd_oled -o 1 -b 15 -g 1 -f 25 -c alsa,hw:Loopback,1

Aug 06 21:06:41 streambox systemd[1]: Starting MPD OLED Display...
Aug 06 21:07:32 streambox systemd[1]: Started MPD OLED Display.
Aug 06 21:07:40 streambox mpd_oled[1311]: Segmentation fault
```
I don't get what's wrong with my setup.
`cava -h`
delivers
`-bash: cava: command not found`
What are the differences on the "updated instructions"? Maybe I missed these details...

antiprism, 2022-08-06:
The updated instructions are just the for the source install instructions to install extra packages. As you are just using the binary package this doesn't affect anything.

I think I have spotted the problem. You are using 'hw' instead of 'plughw' with the loopback device. This also produces the freeze/segfault for me. Instead, try
```
sudo mpd_oled_service_edit -o 1 -b 15 -g 1 -f 25 -c alsa,plughw:Loopback,1
```
Adrian

merlin699, 2022-08-06:
Totally my fault, I saved the service_edit some time ago and missed the "plughw" change...
Now it's working perfectly!
Thanks for your time, I owe you a beer ;-)

antiprism, 2022-08-07:
Great. I'm pleased it's working.

Adrian.

closed 2022-08-07 by antiprism