antiprism/mpd_oled #75: Mpd_oled on a 128x64 RGB matrix

supercrab, 2022-08-06:
Not at issue as such. I’ve ordered one and am gonna port mpd_oled to run with it. It’s coming from China. Just thought you’d like to know 😆

Cheers Mase

antiprism, 2022-08-07:
Hi Mase

Is it like this?

   https://es.aliexpress.com/item/4000219881777.html

Looks cool!

Adrian.

supercrab, 2022-08-07:
Yup, it looks like that one but it’s twice the width so it 128x64 pixels in a single panel. Should be straight forward enough to implement 🤞🏽 

“Low price rgb dot matrix led module 128x64 , p2 led screen module 256mm x 128mm , full color led display p2”

closed 2022-08-07 by supercrab

supercrab, 2022-08-08:
Here is how I plan to wire up the displays and the library I intend to use with it

https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/wiring.md

https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/examples-api-use/minimal-example.cc