antiprism/mpd_oled #76: cp: cannot stat '/var/local/www/currentsong.txt': No such file or directory

supercrab, 2022-08-08:
Hiya!

When I initially run mpd_oled I get the error message, my plugin picks it up in `stderr` and thinks `mpd_oled` has failed.  Is it possible to destroy this error message like this?

main.c: 609
```
	system("cp /var/local/www/currentsong.txt /tmp/currentsong_before_sleep.txt > /dev/null 2>&1");
    usleep(opts.cava_start_delay*1000000);
	system("cp /var/local/www/currentsong.txt /tmp/currentsong_after_sleep.txt > /dev/null 2>&1");
```

cheers
Mase

antiprism, 2022-08-09:
Hi Mase

You can delete the two `system()` lines. They were added in to help monitor an issue were it was sometimes difficult to determine the Moode playback state, but are no longer needed. I will remove them from mpd_oled.

Adrian.

antiprism, 2022-08-09:
I will also remove the associated (undocumented) `-Z` option

   https://github.com/antiprism/mpd_oled/issues/67#issuecomment-839991454

supercrab, 2022-08-09:
Thanks Adrian! 😎👍🏽

supercrab, 2022-08-10:
I'll reopen so I know when you've done it so I can test it!

antiprism, 2022-08-11:
Hi Mase

I have updated the repository with this change, and also made a new binary package (only tested on Moode).

If you are including a binary of cava then the plugin is limited to the systems that the cava binary will run on, in which case you could just install the mpd_oled binary package, then nothing needs to be built (run `mpd_oled_service_uninstall` after installing in order to remove the unwanted mpd_oled systemd service).

Adrian.

supercrab, 2022-08-11:
Thanks, I gave it a quick test and running `mpd_oled -o 3` causes an error 😥

` -bash: /usr/bin/mpd_oled: cannot execute binary file: Exec format error`

here is my install file for info

```
# Install binary of mpd_oled
wget -N http://pitastic.com/mpd_oled/packages/mpd_oled_volumio_install_latest.sh
sudo bash mpd_oled_volumio_install_latest.sh

# Remove mpd_oled service
sudo mpd_oled_service_uninstall
```

antiprism, 2022-08-11:
I built the package on the latest Moode, which is now based on debian bullseye, and so ahead of Volumio which is based on buster. I'll build the package on Volumio instead, and see if it runs on Moode.

supercrab, 2022-08-11:
Ahah!  What a pain in the bum bum!  I didn't even know this could happen - I see what you mean about the system now!

supercrab, 2022-08-11:
it does say `http://pitastic.com/mpd_oled/packages/mpd_oled_volumio_install_latest.sh`  maybe another one for Moode? 🤣🤣

antiprism, 2022-08-11:
The installer script does some different things depending on the player (although that may no longer be necessary with the move to Volumio 3, I'll check), but the binary package can be the same. Unfortunately, a change I made to mpd_oled requires a change to my package builder script, so I am looking at that at the moment.

supercrab, 2022-08-11:
Fair dos! No rush like. Nothing is ever easy is it? 😅

antiprism, 2022-08-11:
Ok, I have hopefully fixed the binary install.

I built the binary package on Volumio 3, and it is working there, but I haven't checked cava with alsa input. The Moode install commands work fine on Volumio 3, but I haven't made any changes relating to this. Also, the Volumio source build may require the libasound2-dev package to be installed, or there will be no alsa support in cava.

The binary package works fine on Moode, and I checked cava with alsa input.

supercrab, 2022-08-11:
@antiprism I've just tested it and everything works!  CAVA also seems to be working fine as well, as far as I can tell!

I'll add your name to the list of authors of the plugins because none of this would have been possible with `mpd_oled` 😯

Cheers

closed 2022-08-11 by supercrab

antiprism, 2022-08-12:
Great! Thanks!

supercrab, 2022-08-12:
I'm gonna have another look at using `mpd_oled` as a service.  I've just noticed the bash scripts to edit the service `sudo mpd_oled_service_edit -o 6 -b 10 -g 1 -f 20`

antiprism, 2022-08-12:
As Volumio will be managing setting up the alsa configuration, you may need to check for its existence in the launch test script (mpd_oled_launch_test), unless this is guaranteed to be set up before mpd is started (which is the current launch test on Volumio).

supercrab, 2022-08-12:
I looked at `mpd_oled_launch_test` and it calls `ps -C mpd`.   I don't know how to check for the alsa configuration but the plugin now comes with an ALSA config file which I assume does all the routing.  Balbuze made these changes so I don't know what is actually happening in the architecture.

antiprism, 2022-08-13:
The potential issue is that cava will fail to start if it tries to access the alsa configuration before it is applied (the configuration file created, however this happens). However, mpd_oled delays running cava until playback has started, and I imagine playback cannot start until Volumio has fully completed its startup, hopefully meaning that the alsa configuration is guaranteed to be available.

supercrab, 2022-08-13:
You've lost me although that is not hard to do :(  Why would CAVA access the alsa config before it's applied?  Currently, the plugin generates the fifo, updates the alsa config and starts `mpd_oled`.  I pass the fifo in as a parameter `mpd_oled  -o 3 -b 16 -g 1 -f 60 -s 8,5 -C 0 -P p -c fifo,/tmp/mpdoledfifo -B 1 -r 25 -D 24 -S 0`

antiprism, 2022-08-13:
I was referring to accessing an alsa plugin device, but I believe there is the same potential issue with the fifo created by the alsa configuration. When is the fifo created? What happens if cava runs and the fifo has not been created? However, as mpd_oled will not run cava until playback has been detected (because of the potential for issues) there should hopefully be no problem.

supercrab, 2022-08-13:
> I was referring to accessing an alsa plugin device, but I believe there is the same potential issue with the fifo created by the alsa configuration. When is the fifo created? What happens if cava [...]

The fifo is created by the plugin itself on startup with this command `/usr/bin/mkfifo -m 646 /tmp/mpdoledfifo`.

I have no idea, how or why it works :(

antiprism, 2022-08-13:
The comment might be saying that because you can't edit the systemd service file, instead of a including a command to run mpd_oled the service file could run a script in /tmp which includes the command to run mpd_oled. You can write the script that is in /tmp and therefore add the mpd_oled options that will be used to this script.

antiprism, 2022-08-13:
So long as the fifo is guaranteed to exist before playback occurs (it probably is) then there should be no issue with mpd_oled/cava.

supercrab, 2022-08-13:
> 

Right!  I understand now!  I’ll give that a try!  Thanks!

supercrab, 2022-08-13:
> Right! I understand now! I’ll give that a try! Thanks!

Oh, if I can’t edit the systemd service file, how can I point it to a script in /tmp 😫

antiprism, 2022-08-14:
I don't know, as I don't really know anything about Volumio plugins.

Why can the following example use sudo https://developers.volumio.com/plugins/index-js#using-daemons , but you can't run mpd_oled_service_edit using sudo? Are different privileges available during different plugin actions?

Can a plugin have a dependency on a repo package, which will then be installed by Volumio during the plugin install? If this is the case, you could submit the mpd_oled debian package for inclusion in the Volumio repo, with the service file modified to run appropriately named scripts in /tmp (I could prepare and maintain this binary package). Then you don't need any special privilges for installing the service file or setting the mpd_oled options.

Not related, but looking here https://docs.google.com/spreadsheets/d/1eRl7ZlMUjOuWTXcSjBgFmO9RI8a3ZJ1U10pi1CWtWy0/edit#gid=0 it seems like the plugin will not be accepted into Volumio if it does not use prebuilt binaries: "No, plugin shall not compile anything on user device".

supercrab, 2022-08-15:
I have managed to get it working by creating a service that points to /tmp/plug_service.sh and then I create said script with a shebang and chmod it and that seems to work fine - just as they described!  Good news.

I think I have solved the problem around using mpd_oled with a service so I don't need to do :)

The plugin doesn't compile anything any more + the plugin doesn't need to use exec.

I think the plugin wouldn't have passed but now it should do!

Thanks

antiprism, 2022-08-15:
Great!