antiprism/mpd_oled #77: Some potentially useful tips for defining the mpd_oled service for Volumio

supercrab, 2022-11-01:
Hi

I've changed the way the Volumio plugin starts `mpd_oled`.  This might of use to you in your scripts but might not - it might reduce the need for that do loop for waiting for mpd.  I thought I'd create an issue so you don't miss it.  Anyway, the plugin seems to work as it should so it might actually get released soon!

The trick is to add these lines to the service:

```
After=network.target sound.target mpd.service
Requires=mpd.service
```

e.g. from my install script (in my script `/tmp/mpd_oled_plugin.sh` just called `mpd_oled`)

```
# create mpd_oled plugin service
service="mpd_oled_plugin"

tmp_file_name="/tmp/$service.service"
tmp_file_contents="[Unit]
Description=MPD OLED Plugin
After=network.target sound.target mpd.service
Requires=mpd.service

[Service]
ExecStart=/tmp/mpd_oled_plugin.sh

[Install]
WantedBy=multi-user.target"

echo "$tmp_file_contents" > $tmp_file_name
```

antiprism, 2022-11-01:
Hi Mase

Thanks for feeding back here.

On Volumio this is a probably a technically better solution, but Moode will still nead the launch test script, and other Player OSs may also need a special check, which is eaily added to the script. I could probably add the "After=mpd.service" to the service file for all systems (and have an automatic pass for the Volumio mpd_oled lauch test), as querying mpd is the default for unsupported systems and so should be running, and mpd is known to be running on supported systems. However, I won't make this change to the current mpd_oled while the existing solution is working.

I'll leave the issue open for a bit in case there are any other comments.

Adrian.

supercrab, 2022-11-01:
Cool, I wasn’t sure if it’s useful or not but definitely a case of it ain’t broke don’t fix it 😂 

Cheers Ad! 👊🏽

Mase

supercrab, 2022-11-05:
sound.target was removed because it causes audio issues

supercrab, 2022-11-05:
Closing as you got the message ☺️

closed 2022-11-05 by supercrab