antiprism/mpd_oled #79: 128x32 Monochrome OLED

CoolTechProjects, 2022-12-15:
How to change the OLED resolution? I've got a 128x32 Monochrome OLED (https://www.adafruit.com/product/3527).
It is working fine on Adafruit 2IC type, but the lines are squizzed.

antiprism, 2022-12-15:
Hi

Your screen is unsupported by mpd_oled, but is supported by the included driver. If you want to use it with mpd_oled you would need to enable screen type -o 2

https://github.com/antiprism/mpd_oled/blob/8299f2bddbcffa6de3f37e97719d62139ee9295b/src/ArduiPi_OLED_lib.h#L49
https://github.com/antiprism/mpd_oled/blob/8299f2bddbcffa6de3f37e97719d62139ee9295b/src/main.cpp#L198

And also customise the display code so the display elements appear in the smaller area

https://github.com/antiprism/mpd_oled/blob/master/doc/FAQ.md#what-code-do-i-change-to-customise-the-display

I am not considering any new development for this version of mpd_oled, and so will not be adding support for your screen. There is, however, a development version of mpd_oled, that may one day support your display

https://github.com/antiprism/mpd_oled_dev
   

Adrian.

CoolTechProjects, 2022-12-15:
Thanks a lot!

closed 2023-03-16 by antiprism