antiprism/mpd_oled #80: Improve framerate by changing image data transmission!

supercrab, 2022-12-23:
I know this branch is maintenance mode but I've discovered something interesting.

I've found during some tests of my cheapo I2C OLED that you can send it all the image data in one go!   Am pretty sure most screens support this to make update faster by reducing the amount of data transferred.

In `ArduiPi_OLED.cpp:914` a buffer array is created with 17 elements `char buff[17]` and the image data is sent in 16 data bytes.  I reckon sending the whole array in one go will improve frame rate.

ArduiPi_OLED.cpp line 914 - 943
```
    char buff[1025];
    uint8_t x;

    // Setup D/C to switch to data mode
    buff[0] = SSD_Data_Mode;

    if (oled_type == OLED_SH1106_I2C_128x64) {
      for (uint8_t k = 0; k < 8; k++) {
        sendCommand(0xB0 + k); // set page addressSSD_Data_Mode;
        sendCommand(0x02);     // set lower column address
        sendCommand(0x10);     // set higher column address
      }
    }

    // loop trough all OLED buffer
    for (i = 0; i < oled_buff_size; i++) {
      buff[i + 1] = buff[i];
    }

    fastI2Cwrite(buff, 1025);
```

 I will report back when I test this on a pi.

antiprism, 2022-12-23:
Hi Mase

I don't know how efficient the code is, as it is Arduino code that was converted to work on the Raspberry Pi. It will be interesting to see what results you get.

Adrian.

supercrab, 2022-12-23:
There are 124*64 = 8192 pixels to send, which is 1024 bytes.  
The current code would send 16 data bytes + 1 control byte 64 times by calling `fastI2Cwrite()`.  
The new code would send 1 control byte + 1024 data bytes with a single call to `fastI2Cwrite()`.
Will let ya know!