antiprism/mpd_oled #81: Unable to use hifiberry digi pro DAC with mpdoled

arniecommits, 2023-02-20:
Hi,

I have been trying to troubleshoot, this and changed gpio pins and i2c bus configuration. What seems to happen is that when using mpd oled volumio comes up with an error saying Alsa resource busy, I think the writes to the oled is failing and it causes the whole thing to freeze. The screen itself works fine. I am able to use the same screen with pydpiper without any issues. Please help.

antiprism, 2023-02-20:
Hi

If you are getting ALSA errors then I would guess that you are using the Volumio 3 mpd_oled plugin, in which case the initial place to ask for support for the plugin is probably

   https://community.volumio.org/t/plugin-mpd-oled-installation-configuration-plugin/44823

The error relates to making a copy of the audio for the spectrum display, which is why you don't see a similar issue when using pydpiper.

Adrian.

supercrab, 2023-02-20:
If you’re using the plungin, I’d suggest removing it and installing mpd_oled as standalone. I don’t think ALSA is robust enough.

arniecommits, 2023-02-20:
Thanks both, removing the plugin has certainly helped, is it normal for the mpd_oled process to consume 90% of the CPU on a RPI 4 ?

antiprism, 2023-02-20:
It is not normal to consume so much CPU. On a Pi Zero while playing a radio stream with mpd_oled running CPU usage is around 90%. How many spectrum bars are specified and how many frames per second?

arniecommits, 2023-02-20:
16 bars and 20 frames, I set the i2c baudrate as recommended, I am playing fairly high resolution flac files , does it have an impact ?

antiprism, 2023-02-20:
The mpd_oled settings you are using are fine. Is the high CPU usage when music is playing, or all the time? Could you restart the Pi, and if mpd_oled is still using a lot of CPU then post the output of
```
top -b -n 1
```
Adrian.

supercrab, 2023-02-20:
What happens if you play low quality web radio? 🤔

arniecommits, 2023-02-20:
This is idle usage 
[image]
While playing 
[image]

arniecommits, 2023-02-20:
I just checked, its happening irrespective of stream quality

arniecommits, 2023-02-20:
oops sorry about that , i2c-1 is being used by the DAC HAT, while I could add it as a slave, I avoided the whole thing, and used /dev/i2c-4 instead like this 
bash-5.0# cat /boot/userconfig.txt 
# Add your custom config.txt options to this file, which will be preserved during updates
dtoverlay=i2c-gpio,bus=4,i2c_gpio_sda=16,i2c_gpio_scl=25
dtparam=i2c_arm_baudrate=800000

MPD Oled Parameters
ExecStart=/usr/bin/mpd_oled  -o 6 -b 16 -g 1 -f 20 -B 4 -a 3c

antiprism, 2023-02-20:
I believe that the i2c-gpio driver is implementing the I2C protocol in software, and hence running on the CPU. If you use /dev/i2c-1 then the protocol is implemented in hardware and you will have a lower CPU usage.

Adrian.

arniecommits, 2023-02-20:
Thanks! I rewired to i2c-1 and everything is working well now, CPU usage is back down to 10%, you guys are awesome! thank you so much for the help much appreciated!

closed 2023-02-20 by arniecommits