antiprism/mpd_oled #84: MPD_oled over MoOde audio rel 8.3.2

andromeda2791, 2023-05-11:
Hi Adrian
with the new rel. of MoOde 8.3.2
the your magnificent code 4 oled displays don't work, can you have some suggestion?
the oled remain freezed and no spectral bars is displayed...
tnks 4 Ur work

antiprism, 2023-05-12:
Hi

Thanks for reporting the issue. There was a change to the Moode startup code in 8.3.2 which affects the mpd_oled startup (and vice versa). It is discussed here

   https://moodeaudio.org/forum/showthread.php?tid=5556

I hope to find some time to update mpd_oled over the weekend.

Adrian.

andromeda2791, 2023-05-12:
Tnks Adrian just 2 word
I'm a moderator of a little community in FB (now with over 1150 people...)
where I've proposed your solution for make the architecture of HiFi RPi
based DAC more interesting and aesthetic... On Old MoOde revs ...work very
well and was very appreciated with a lot of positive feedback over the
appearance that your code gave to the final product ...
Eventually could be more fascinating if do possible to insert a start
jingle Logo/animation and if can be modified the char typology ...
If can be updated with new features.. you must have much fans!
Thank 4 your work
Eug.

antiprism, 2023-05-14:
Hi Eug.

Thank you for your comments and suggestions!

I have updated the repository and the binary packages with a fix for this issue.

This version of mpd_oled is in maintenance mode, and I do not plan to add any new features. I did, however, start a new version of mpd_oled for implementing new features

   https://github.com/antiprism/mpd_oled_dev

It uses a different library for controllers and graphics, and supports more types of screen. I also made a font with international languages to use for display, although have not yet added this into the program. However, as I cannot easily buy a Pi at the moment I have put development of this new version on hold for now.

Adrian.

andromeda2791, 2023-05-14:
Great Adrian!
go to TRY immediately!!

andromeda2791, 2023-05-15:
unfortunately on my installation not work propely during reproduction remain lighted ononly time label volume , tx-rx  icone and the base line of speactrum graph... no other info or moving over oled

antiprism, 2023-05-16:
Are you installing mpd_oled from source, or using the deb package?

Was there previously a version of mpd_oled on this machine, and was it installed the same way?

Please run the following command while playing music, and post the output

```sudo systemctl status mpd_oled | cat```

Also, post the contents of the file `/etc/systemd/system/mpd_oled.service`

Adrian.

andromeda2791, 2023-05-17:
Hi Adrian tnks
the installation used was by source 
the status mpd_oled done the result ...

● mpd_oled.service - MPD OLED Display
     Loaded: loaded (/etc/systemd/system/mpd_oled.service; enabled; vendor prese                                                                                            t: enabled)
     Active: active (running) since Tue 2023-05-16 00:19:46 CEST; 1 day 17h ago
   Main PID: 449 (mpd_oled)
      Tasks: 2 (limit: 9293)
        CPU: 4.198s
     CGroup: /system.slice/mpd_oled.service
             └─449 /usr/local/bin/mpd_oled -o 7 -b 32 -g 1 -f 25 -s 25.0 5.0 -c                                                                                             alsa,plughw:Loopback,1

May 16 00:19:46 MyRASpiDAC systemd[1]: Started MPD OLED Display.

(that mean ..all appear to be ok...)
and the mpd_oled.service is configured as
 

[Unit]
Description=MPD OLED Display

[Service]
# ExecStart=/usr/local/bin/mpd_oled_launch /usr/local/bin/mpd_oled -o 0 -b 21 -g 1 -f 15
ExecStart=/usr/local/bin/mpd_oled -o 7 -b 32 -g 1 -f 25 -s 25.0,5.0 -c alsa,plughw:Loopback,1
[Install]
WantedBy=multi-user.target

antiprism, 2023-05-17:
Thanks. There is an issue with the service file, as it has not been updated by the new install (which will not overwrite a current service file). Try running
```
sudo mpd_oled_service_uninstall
sudo mpd_oled_service_install
sudo mpd_oled_service_edit -o 7 -b 32 -g 1 -f 25 -s 25.0,5.0 -c alsa,plughw:Loopback,1
```
This should remove the service file, reinstall it, then add back in the options from your current service file.

If mpd_oled is not running with a working spectrum after this, please post the output of of the status command and the contents of the service file again.

Adrian.

andromeda2791, 2023-05-17:
great! .... workin now

antiprism, 2023-05-17:
Great. I'll post to the Moode forum about updating. Thanks for the feedback!

andromeda2791, 2023-05-17:
my pleasure for debugging your work ... 
You update all code in github surce and also for deb package?

antiprism, 2023-05-17:
Yes. I made deb package iwith the current source.

Changing the font or what is displayed  involves modifying the C++ code.

andromeda2791, 2023-05-17:
can give some indication how to that feature is implemented and where is defined in the code
I'm courious if can see some in the code

antiprism, 2023-05-17:
In main.cpp there are two simple functions that draw the layout

draw_clock(): draws the stop screen draw_spect_display(): draws the play/pause screen

They run the component display functions, so you can see the names of the functions you want to change.

andromeda2791, 2023-05-17:
I've rate U (+1) over moode forum (my nick is AndromedaStrain)

andromeda2791, 2024-10-02:
hi Adrian how do you do?
recently I've see a new rev of MoOde 9.1.0 but the mpd feature oled goes
lost.
have you some news thinks that need to know for make working the OLED...?

tnks 4 Ur support

Eug

antiprism, 2024-10-03:
Hi Eug

You are using `mpd_oled -o 7`, and so have an SPI screen. It appears that Moode 9.1 is now based on Raspberry Pi OS Bookworm, which I believe removess a particular way (sysfs) that software can access GPIO pins which mpd_oled uses for SPI screens (not I2C screens).

I won't be fixing this, sorry, as I am no longer actively developing mpd_oled, and the issue requires more than just a quick fix.

Perhaps you could look at whether your screen can be configured to use I2C.

Adrian.

antiprism, 2024-10-03:
Hi Eug

I see that I have had success running mpd_oled with an SPI screen on Moode 9.0, which is also based on Bookworm

   https://moodeaudio.org/forum/showthread.php?tid=155&pid=56820#pid56820

However, a kernel update may now have removed the required interface. To check, what result do you get when you run

```ls /dev/gpiomem```

Adrian.