antiprism/mpd_oled #85: A mpd_oled is crashing in idle/clock mode  after switching to 64bit.

kan-ibal, 2023-08-15:
I have upgraded my rAudio based on Arch from 32bit to 64bit and cannot make mpd_oled to work while in idle/clock mode. During a playback mpd_oled is working ok.

```
systemctl status mpd_oled 

mpd_oled.service - MPD OLED Display
     Loaded: loaded (/etc/systemd/system/mpd_oled.service; disabled; preset: disabled)
     Active: failed (Result: core-dump) since Tue 2023-08-15 12:37:21 CEST; 3s ago
   Duration: 1min 32.529s
    Process: 4662 ExecStart=/usr/local/bin/mpd_oled -a 3c -o 3 -b 10 -g 1
   Main PID: 4662 (code=dumped, signal=ABRT)
        CPU: 32.550s

Aug 15 12:35:47 rAudio systemd[1]: Started MPD OLED Display.
Aug 15 12:37:20 rAudio systemd[1]: Stopping MPD OLED Display...
Aug 15 12:37:21 rAudio systemd[1]: mpd_oled.service: Main process exited, code=dumped, status=6/ABRT
Aug 15 12:37:21 rAudio systemd[1]: mpd_oled.service: Failed with result 'core-dump'.
Aug 15 12:37:21 rAudio systemd[1]: Stopped MPD OLED Display.
Aug 15 12:37:21 rAudio systemd[1]: mpd_oled.service: Consumed 32.550s CPU time.
```
and

```
journalctl -r

Aug 15 12:37:21 rAudio systemd[1]: systemd-coredump@3-6487-0.service: Deactivated successfully.
Aug 15 12:37:21 rAudio systemd[1]: mpd_oled.service: Consumed 32.550s CPU time.
Aug 15 12:37:21 rAudio systemd[1]: Stopped MPD OLED Display.
Aug 15 12:37:21 rAudio systemd[1]: mpd_oled.service: Failed with result 'core-dump'.
Aug 15 12:37:21 rAudio systemd[1]: mpd_oled.service: Main process exited, code=dumped, status=6/ABRT
Aug 15 12:37:21 rAudio systemd-coredump[6492]: [LNK] Process 4662 (mpd_oled.new) of user 0 dumped core.
                                               
                                               Stack trace of thread 4662:
                                               #0  0x0000007f9eef2790 n/a (libc.so.6 + 0x82790)
                                               #1  0x0000007f9eeab6fc raise (libc.so.6 + 0x3b6fc)
                                               #2  0x0000007f9ee978b0 abort (libc.so.6 + 0x278b0)
                                               #3  0x000000559175b03c n/a (/usr/local/bin/mpd_oled + 0x2b03c)
                                               #4  0x000000559175b03c n/a (/usr/local/bin/mpd_oled + 0x2b03c)
                                               #5  0x0000007f9f1917ac n/a (linux-vdso.so.1 + 0x7ac)
                                               #6  0x0000007f9ef2484c clock_nanosleep (libc.so.6 + 0xb484c)
                                               #7  0x000000559187c648 n/a (/home/.local/bin/mpd_oled.new + 0x14c648)
                                               ELF object binary architecture: AARCH64
Aug 15 12:37:20 rAudio systemd[1]: Started Process Core Dump (PID 6487/UID 0).
Aug 15 12:37:20 rAudio systemctl[6481]: See system logs and 'systemctl status mpd_oled.service' for details.
Aug 15 12:37:20 rAudio systemctl[6481]: Failed to stop mpd_oled.service: Access denied
Aug 15 12:37:20 rAudio systemd[1]: Stopping MPD OLED Display...
```

I'm not sure it is a mpd_oled issue and for me it looks like mpd_oled is crashing because systemd is stopping its service. 
But why? The same setup is working in 32bit system.

antiprism, 2023-08-15:
Hi

Thanks for reporting the issue. Just to check, you have built a 64 bit mpd_oled binary from source on the sytem you are running the binary on. Were there any warnings printed when compiling? If you aren't sure, could you rebuild mpd_oled and review the output for any warnings or errors.

Adrian.

kan-ibal, 2023-08-15:
Yes, I have built it on the same system I'm running the binary on. There were no warnings when compiling. I have to add that when I run binary in terminal from command line it works as it suppose to.

antiprism, 2023-08-15:
The issue seems strange.

Just to confirm
* not playing music and running mpd_oled from the command line works fine, starting and stopping the music works fine
* playing music and running mpd_oled from the command line works fine, stopping and starting the music works fine
* not playing music and starting the mpd_oled service causes a crash
* playing music and starting the mpd_oled service does not cause a crash, stopping the music causes a crash

The backtrace contains the line

   #7  0x000000559187c648 n/a (/home/.local/bin/mpd_oled.new + 0x14c648)

Do you know anything about "/home/.local/bin/mpd_oled.new"?

Could you post the contents of: /etc/systemd/system/mpd_oled.service

Adrian.

kan-ibal, 2023-08-18:
> * not playing music and running mpd_oled from the command line works fine, starting and stopping the music works fine

yes

> * playing music and running mpd_oled from the command line works fine, stopping and starting the music works fine

yes (as above)

> * not playing music and starting the mpd_oled service causes a crash
> * playing music and starting the mpd_oled service does not cause a crash, stopping the music causes a crash

yes - stopping or pausing(mpc client)

> 
> The backtrace contains the line
> #7 0x000000559187c648 n/a (/home/.local/bin/mpd_oled.new + 0x14c648)
> Do you know anything about "/home/.local/bin/mpd_oled.new"?

Yes, my mistake. Should be as above, /usr/local/bin. That is from my other test attempt.

> Could you post the contents of: /etc/systemd/system/mpd_oled.service

Now is:
```
[Unit]
Description=MPD OLED Display

[Service]
ExecStart=/usr/local/bin/mpd_oled -k -a 3c -o 3 -b 10 -g 1 -f 25 -c fifo,/tmp/mpd.fifo

[Install]
WantedBy=multi-user.target
```

antiprism, 2023-08-18:
Hi

The service file isn't as expected, as it should start mpd_oled from a launch script that waits for mpd to start. I am not sure if it will help, but it would make sense to have the service file setup like a new install for testing. If you are not running the repository version of mpd_oled then please install that version instead. In *all* cases (even if you reinstall), reset the service file with
```
sudo mpd_oled_service_uninstall
sudo mpd_oled_service_install
sudo mpd_oled_service_edit -k -a 3c -o 3 -b 10 -g 1 -f 25 -c fifo,/tmp/mpd.fifo
```

List the contents of `/etc/systemd/system/mpd_oled.service` and check it is now running `mpd_oled_launch` instead of `mpd_oled`

Adrian.

kan-ibal, 2023-08-18:
I don't think it will help as well. Yes, I admit that my  service file is custom  but for testing  purposes I followed your path but mpd_oled_service_edit gives me:
```
rAudio:~/mpd_oled/scripts  #  ./mpd_oled_service_edit -k -a 3c -o 3 -b 10 -g 1 -f 25 -c fifo,/tmp/mpd.fifo
setting service file to run: mpd_oled '-k -a 3c -o 3 -b 10 -g 1 -f 25 -c fifo,/tmp/mpd.fifo'
sed: -e expression #1, char 79: unknown option to `s'
```
using  manual setup:
```
[Unit]
Description=MPD OLED Display

[Service]
ExecStart=/usr/local/bin/mpd_oled_launch /usr/local/bin/mpd_oled -k -a 3c -o 3 -b 10 -g 1 -f 25 -c fifo,/tmp/mp
d.fifo -p mpd

[Install]
WantedBy=multi-user.target

```
doesn't change anything.

antiprism, 2023-08-19:
Hi

I tracked down the `sed` error, and will fix it, thanks.

It is difficult to suggest more things to try, but maybe it is related to the system user that is used to run the mpd_oled program when the service starts it.

I'll see if I can install here and take a look. Where can I download the rAudio image are you using? Are there any other instructions to set up the system as you have done?

Adrian.

kan-ibal, 2023-08-19:
Full image you can find [here](https://github.com/rern/rAudio/releases/tag/i20230620) and update via rAudio GUI to the latest. There is no additional system setup I have that may be applicable for that mpd_oled issue.
For my display, in /boot/config.txt  I have following settings:
```
dtparam=i2c_arm=on
dtparam=i2c_arm_baudrate=800000
```
and in /etc/modules-load.d/mpd_oled.conf
`i2c-dev`

How to set wi-fi you can find in [readme](https://github.com/rern/rAudio/blob/i20230620/README.md).

kan-ibal, 2025-03-20:
I have recompiled mpd_oled and cava once again and it now works as it supposed to. Fr me it looks like the culprit was a cava.

antiprism, 2025-03-21:
Thanks for the update. I pleased it is working now!

Adrian.