antiprism/mpd_oled #87: Spectrum Analyzer  after some time in stop

erl4ever, 2024-05-08:
Hello, I use MPD OLED on Volumio even before the plugin version.

I'm using the latest version of Volumio as of this date, and when the player enters STOP, after some time the Spectrum Analyzer stops working. It's necessary to disable and re-enable the plugin to get back to normal.

Other display information continues to work normally.

Volumio version: 3.661
Plugin version: 1.12
Raspberry Pi 3B

Thank you!

antiprism, 2024-05-08:
Hi

If you have installed mpd_oled from the Volumio plugin store then I recommend asking for help at:

   https://community.volumio.com/t/plugin-mpd-oled-installation-configuration-plugin/44823

If I understand, you are seeing the clock screen, but when you play music again the display does not change from the clock screen. If the clock time is not changing on the display then mpd_oled has crashed. If this is the case, then maybe the volumio log will include the error message associated with the crash.

Adrian.

erl4ever, 2024-05-09:
Adrian, thanks for the quick response.

I installed through the Volumio store.
When starting Volumio everything works, the clock, the music information, bitrate, spectrum analyzer, etc. When I pause the music, back from the break everything works. However, when I put the player in "stop", after a few minutes stopped when returning with "Play" everything works, except the bars of the spectrum analyzer. It is necessary to restart the plugin to run to work.

antiprism, 2024-05-09:
From your description, it sounds like the issue relates to the audio copy made for calculating the spectrum, which is configured within the Volumio plugin (and I don't know all the details). I recommend reporting the issue on the thread I mentioned in the Volumio forum.

Adrian.

supercrab, 2024-05-09:
Adrian is correct; there seems to be an issue with the audio stream. Unfortunately, my knowledge of audio streams is limited, so I'm unsure how to resolve it. The plugin acts as a basic facade, merely setting up mpd_oled as a service, configuring it, and managing its start and stop functions. Despite my efforts, the plugin has never functioned properly for me on Volumio 3. Therefore, I suggest uninstalling it and instead using and installing mpd_oled as a standalone application.

Interestingly, when the plugin didn't utilize the audio copy feature in Volumio 2, it worked flawlessly. 😫

supercrab, 2024-05-09:
I reviewed the code and found that the plugin utilizes an ALSA configuration file to direct the audio to 'mpd_oled'. There doesn't seem to be any apparent reason for the issue, aside from the possibility of Volumio behaving unexpectedly with the stream.