antiprism/mpd_oled #89: Just for info: works with moOde 9 if installed using source

merlin699, 2024-08-20:
Just to let you know: after failing installing using the binary package, I successfully installed MPD OLED on my Rpi3b+ with SPI display and moOde Version 9 using the "from source" method for moOde 8:
https://github.com/antiprism/mpd_oled/blob/master/doc/install_moode8_source.md

ROCH-23, 2025-05-01:
Привет! Я не могу загрузить дисплеи,команда не проходит
cd ..   # if you are still in the cava source directory
git clone https://github.com/antiprism/mpd_oled
cd mpd_oled
./bootstrap
CPPFLAGS="-W -Wall -Wno-psabi" ./configure --prefix=/usr/local
make
sudo make install-strip

pi@moOde:/home $ pi@moOde:/home $ sudo make install-stripcd ..   # if you are still in the cava source directory
-bash: pi@moOde:/home: No such file or directory
pi@moOde:/home $ git clone https://github.com/antiprism/mpd_oled
-bash: git: command not found
pi@moOde:/home $ cd mpd_oled
-bash: cd: mpd_oled: No such file or directory
pi@moOde:/home $ ./bootstrap
-bash: ./bootstrap: No such file or directory
pi@moOde:/home $ CPPFLAGS="-W -Wall -Wno-psabi" ./configure --prefix=/usr/local
-bash: ./configure: No such file or directory
pi@moOde:/home $ make
make: *** No targets specified and no makefile found.  Stop.
pi@moOde:/home $ sudo make install-stripmake: *** No rule to make target 'install-stripcd'.  Stop.
pi@moOde:/home $
make: *** No rule to make target 'install-stripmake:'.  Stop.

antiprism, 2025-05-02:
Hi ROCH-23

Please post in English.

There appear to be (at least) two issues:

You should be in /home/pi, and cava and mpd_oled subdirectories, but you appear to have run an extra `cd ..` which has taken you up a level to /home.

It looks like git is not installed, so run
```
sudo apt install git
```
Adrian.

ROCH-23, 2025-05-02:
pi@moOde:/ $ sudo apt install git
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
git is already the newest version (1:2.39.5-0+deb12u2).
0 upgraded, 0 newly installed, 0 to remove and 89 not upgraded.
pi@moOde:/ $ cd ..   # if you are still in the cava source directory
pi@moOde:/ $ git clone https://github.com/antiprism/mpd_oled
fatal: could not create work tree dir 'mpd_oled': Permission denied
pi@moOde:/ $ cd mpd_oled
-bash: cd: mpd_oled: No such file or directory
pi@moOde:/ $ ./bootstrap
-bash: ./bootstrap: No such file or directory
pi@moOde:/ $ CPPFLAGS="-W -Wall -Wno-psabi" ./configure --prefix=/usr/local
-bash: ./configure: No such file or directory
pi@moOde:/ $ make
make: *** No targets specified and no makefile found.  Stop.
pi@moOde:/ $ sudo make install-strip
make: *** No rule to make target 'install-strip'.  Stop.
pi@moOde:/ $

antiprism, 2025-05-02:
Don't run this command
```
pi@moOde:/ $ cd .. # if you are still in the cava source directory
```
Because you are not in the cava source directory

ROCH-23, 2025-05-02:
What commands should I write to load the displays?

antiprism, 2025-05-02:
Carry on with the instructions from here

   https://github.com/antiprism/mpd_oled/blob/master/doc/install_moode8_source.md#build-and-install-mpd_oled

But don't run the first command. (Don't run `cd .. # if you are still in the cava source directory`)

antiprism, 2025-05-02:
And, you should start running commands in the directory /home/pi. Check this before you start by running
```
pwd
```
It should print `/home/pi`

ROCH-23, 2025-05-02:
checking for uint64_t... yes
checking for uint8_t... yes
checking for GNU libc compatible malloc... yes
checking for sys/param.h... yes
checking for getpagesize... yes
checking for working mmap... yes
checking for gettimeofday... yes
checking for memset... yes
checking for mkfifo... yes
checking for munmap... yes
checking for select... yes
checking for setlocale... yes
checking for socket... yes
checking for strerror... yes
checking for strspn... yes
checking for strstr... yes
checking for strtol... yes
checking for strndup... yes
checking LIBMPDCLIENT_DIR... no (using installed libmpdclient)
checking for mpd/client.h... no
configure: error: Install a development version of this library and rerun ./configure
pi@moOde:~/mpd_oled/mpd_oled $ make
make: *** No targets specified and no makefile found.  Stop.
pi@moOde:~/mpd_oled/mpd_oled $ sudo make install-strip

antiprism, 2025-05-02:
```
pi@moOde:~ $ sudo apt update
sudo apt install autoconf make libtool libfftw3-dev libiniparser-dev libmpdclient-dev libi2c-dev i2c-tools lm-sensors git libasound2-dev
```
There is no prompt for the second command, and I see no output for it.  Copy and paste the commands one at a time. It appears that the way you copied the commands you did not run
```
sudo apt install autoconf make libtool libfftw3-dev libiniparser-dev libmpdclient-dev libi2c-dev i2c-tools lm-sensors git libasound2-dev
```

ROCH-23, 2025-05-02:
Adrian, everything worked! The last command helped to solve the problem! I think it is possible to make instructions for installing version 9. Everything works, many thanks to Adrian!

antiprism, 2025-05-02:
That is great! I'm really pleased it worked out.

tobeehh, 2025-08-01:
@antiprism  I followed your instructions, however cava is not running

pi@beosound3:~ $ cava -h
-bash: cava: command not found

the screen works, but doesnt show any title info or spectrum when the raspberry is chosen as  audio player

Host:  beosound3
RPiOS: 12.11 Bookworm 64-bit | Linux: 6.12.25 64-bit
Model: Pi-Zero 2 W 512MB
Audio: Pi HDMI 1

antiprism, 2025-08-01:
Hi tobeehh

If you built cava following the instructions for installing mpd_oled from source then the executable will be called `mpd_oled_cava'. Are you running Moode 9?

Adrian.