antiprism/mpd_oled #90: Output to USB DAC becomes constant when spectrum analyser is working in mpd_oled

silentcossack, 2024-08-27:
Hi, i'm so lost, hoping this is not a bug, but maybe it is.

This is hard to describe, but i'll try. I got Moode 9 running on rpi 4. Everything works great. I installed mpd_oled from source and it works too. If i run the mpd_oled with default cava settings (i.e. not specifying the source), then the spectrum is not working, but the audio output to my usb dac is great, unmodified, with rate matching that of source. So if source is 48KHZ, then output is 48Khz is well. If source is 44.1, then so is the output.

So if I run this:
/usr/local/bin/mpd_oled -o3 -a 3c

Then spectrum does not work, but audio output quality changes with the source, which is perfect.

However, if I run this:
/usr/local/bin/mpd_oled -o3 -a 3c -c alsa,hw:Loopback,1

Then spectrum does work, but the audio output quality is stuck at whatever the first song was. So if the first song was 48kHz, then all subsequent music will be output at that rate, not matching the source. If i stop mpd_oled, guess what, output starts matching the source again. If I start mpd_oled with the -c option, same problem. It is as if whenever mpd_oled (or cava) connects to the loopback, something starts resampling all outputs, including my usb dac output.

I have the loopback option enabled in Moode web interface. I tried running mpd_oled_moode_audio_copy_install - didn't change anything. Below is the output of mpd_oled_moode_audio_copy_report in case it helps.

```
$ sudo mpd_oled_moode_audio_copy_report
===========================================================
Audio Copy Report for mpd_oled
===========================================================
aplay -l
--------
**** List of PLAYBACK Hardware Devices ****
card 0: Lite [E30 II Lite], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: Loopback [Loopback], device 0: Loopback PCM [Loopback PCM]
  Subdevices: 8/8
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
  Subdevice #7: subdevice #7
card 1: Loopback [Loopback], device 1: Loopback PCM [Loopback PCM]
  Subdevices: 8/8
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
  Subdevice #7: subdevice #7
card 2: vc4hdmi1 [vc4-hdmi-1], device 0: MAI PCM i2s-hifi-0 [MAI PCM i2s-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
===========================================================
aplay -L
--------
null
    Discard all samples (playback) or generate zero samples (capture)
default
    Default Audio Device
sysdefault
    Default Audio Device
_audioout
plug_alsaequal
btstream
camilladsp
crossfeed
plug_bs2b
plug_eqfa12p
invpolarity
trx_send
eqfa12p
eqfa12p_and_copy
[... 193 lines cut ...]
coverurl=/coverart.php/NAS%2FStorage%2FMetallica%2F%2072%20Seasons%20%5B288984589%5D%20%5B2023%5D%2F01%20-%20Metallica%20-%2072%20Seasons.flac
track=1
date=2023-04-14
composer=
encoded=FLAC 16/44.1 kHz, 2ch
bitrate=1.156 Mbps
outrate=Not playing
volume=70
mute=0
state=pause
===========================================================
ps ax | grep mpd_oled
---------------------
    441 ?        Dsl    0:18 /usr/local/bin/mpd_oled -o3 -a 3c -c alsa,hw:Loopback,1
   2468 ?        S      0:00 sh -c mpd_oled_cava -p /tmp/cava_config_oc7xPU
   2469 ?        Sl     0:24 mpd_oled_cava -p /tmp/cava_config_oc7xPU
  26795 pts/0    S+     0:00 sudo mpd_oled_moode_audio_copy_report
  26796 pts/1    Ss     0:00 sudo mpd_oled_moode_audio_copy_report
  26797 pts/1    S+     0:00 /bin/bash /usr/local/bin/mpd_oled_moode_audio_copy_report
  26812 pts/1    S+     0:00 grep mpd_oled
===========================================================

```

What could be changing the rate of output when spectrum is running?

antiprism, 2024-08-27:
Hi silentcossack

My understanding is that when cava starts it will set the sample rate at the loopback device to whatever is playing, then, when the audio sample rate should change through playing some different audio, the player finds that the loopback device has a particular rate and so synchronises to that. I believe that the solution would be for mpd_oled to restart cava every time that play resumes after being stopped, but I am no longer developing mpd_oled and so this won't be fixed.

Adrian.

silentcossack, 2024-08-27:
Ok thanks, that makes sense. I did not realise that you are no longer developing this, too bad. I'll just run it without spectrum for now and see if I change the code to get rid of it and fill it with something else.

silentcossack, 2024-08-28:
For anybody who comes across this, I decided to fork this repo and remove the spectrum display. I also enlarged the artist and song text to take up more display space and be visible from further away. You can grab the modified code here:
[https://github.com/silentcossack/mpd_oled](url)

[image]